MKesenheimer/fault-injection-library #2: [question] porting to kicad ?

FabienTregan, 2024-09-28:
Hi,

  I wanted to try a few modifications on the board*, but the files are for Fusion360 for which I do not (and don't plan to) own a license. Would you be interested in porting the project to Kicad ? (I think I can do the job, but have no interest in maintaining a Kicad fork if you prefer Fusion).
  
  
* : mainly the possibility to have different voltages for power and glitching, since the CC2510 I play with gives better results when glitching from the 1.8V regulation pins but still need to be power-cycled through the 3V3 pin and use 3V3 for communicating on the debug interface.

MKesenheimer, 2024-09-29:
Hey Fabien,
if you are willing to do the work, feel free to do so. I am happy for contributions. I do not use Kicad currently, but if you can port the project to this program, I am open to a change since I am not that happy with Fusion360 regarding to PCB prototyping. Maybe this would be the perfect time to switch gears.
To sum up: Thank you for your kind offer to port the project. I would like to maintain the project in Kicad after you ported it ;)

FabienTregan, 2024-09-29:
I could not open the `.fsch` and `.fbrd` files (eagle crashes on my two computers under linux mint). If you have an option in fusion to export the file to a more standard format (eagle 'old' .sch and .brd ?), I could start from your file instead of redoing from scratch. If you can, just attach the files to this issue and I'll give it a try.

MKesenheimer, 2024-09-29:
I added the files `*.brd` and `*.sch` to `schematics`. Let me know if these files formats work.

FabienTregan, 2024-09-29:
After a few crashes they seem to be imported fine ! I need to check all the biding between components in schematic and PCB, but that will be way easyer than redoing everything from scratch, thanks !

FabienTregan, 2024-09-29:
(I'll try sending a first version during next week)

MKesenheimer, 2024-09-29:
That's good news. Looking forward to your PR.

FabienTregan, 2024-09-30:
On component is present twice on the schematic with the same reference (because `x-glitch` is automatically numbered as `x-glitch0` which already existed) so I renamed this way. Can you confirm it is correct ?
[image]
[image]

MKesenheimer, 2024-09-30:
Ah, I see. Yeah, that should be fine I guess.