MKesenheimer/fault-injection-library #5: [discussion] initiating user documentation ?

FabienTregan, 2024-10-13:
Hi,

 would this be the proper time to initiate user documentation ? as a beginner trying to test the project, here are the chapter which would really have helped me getting into the code / project:

- basic architecture (what is running on computer side, on the rp pico, ...)
- pins usage (e.g. difference between `lpglitch`, `hpglitch`, `x_glitch0`, `x_glitch`,  ... far all pins)
- how to run the samples (which firmware / python code to send to the RP Pico, what to run on computer side, ...)

MKesenheimer, 2024-10-13:
Yes, a full documentation is still missing and I am aware of this. Do you have any recommendations for a documentation framework?

FabienTregan, 2024-10-13:
Since I like Rust, I use mdbook as a basic static website generator from markdown files. Output with default settings looks like this : https://tregan.fr/NoCV/ 

The "extra" work is minimal (just provide a markdown file for the table of content on the left), and the files can probably be reused in lot of static generators that takes markdown as an imput.

There is an extension to generate diagrams from plantuml, which is a bit less common but allow text bases, human readable / maintainable diagrams (great for git + usable without installing anything). Plantuml looks like this (see example for sequence diagram : https://plantuml.com/sequence-diagram )

I have gitlab CI script to generate the static file and upload to my webserver when I push, this is probably easy to do and already documented for github actions.

I don't have any suggestion to document the python code / api since I've only done minimal work with python.

FabienTregan, 2024-10-13:
There is this indication that I don't know where to position (it was over a lot of things)
[image]

MKesenheimer, 2024-10-18:
Documentation of the first classes and methods is available under [https://fault-injection-library.readthedocs.io/](https://fault-injection-library.readthedocs.io/).
I leave this issue open until the documentation is in a "complete-enough" state.

MKesenheimer, 2024-11-01:
Documentation of important functions and classes finished. This still needs to be improved or updated in some places, but documentation is generally complete.

closed 2024-11-01 by MKesenheimer