MKesenheimer/fault-injection-library #12: reset_target into the PicoGlitcher object is missing

inode-, 2025-03-12:
In the PicoGlitcher code the "release_reset" is implemented, but the "reset_target()" is not.

https://github.com/MKesenheimer/fault-injection-library/blob/b5ec99b4813cef692553d9ec373e20b5441a9b4e/findus/findus.py#L829

it can be accessed by using the "self.pico_glitcher" internal object, but it's not the best.

The idea is to:

```
reset_target()
arm()
sleep()
release_reset()
```

MKesenheimer, 2025-03-12:
use the function `reset()` instead. That should be exactly what you want.

MKesenheimer, 2025-03-12:
https://fault-injection-library.readthedocs.io/en/latest/picoglitcher/#findus.firmware.PicoGlitcher.PicoGlitcher.reset

inode-, 2025-03-12:
hmm, if I'm not wrong the reset it blocking: https://github.com/MKesenheimer/fault-injection-library/blob/b5ec99b4813cef692553d9ec373e20b5441a9b4e/findus/firmware/PicoGlitcher.py#L578

I need to arm before the release of the reset.

MKesenheimer, 2025-03-12:
Yes, reset is blocking.

Why does 

```
self.glitcher.arm(delay, length)
self.glitcher.reset(0.01)
```

not work in your case?

This arms the Pico Glitcher (waits for the trigger line to rise), and pulls the reset line down for 0.01 seconds. Afterwards the reset line rises and the Pico Glitcher triggers on the rising edge of the reset line.
You must also connect the reset with the trigger line.

inode-, 2025-03-14:
It was related to: https://github.com/MKesenheimer/fault-injection-library/issues/16

MKesenheimer, 2025-03-14:
Is this still a problem? Or does the use of dead_time and the external trigger inputs EXT1 and EXT2 fix this issue?

MKesenheimer, 2025-03-16:
Replaced `reset` by `reset_target` in the latest release. This should be easier to understand what this function does. The function `reset` is still left in for downward compatibility. You can still use `reset` to reset your target.

Also the original function `reset_target` was replaced by `initiate_reset` for consistency.

Closed since the original question seems to be resolved. If not, please re-open.

closed 2025-03-16 by MKesenheimer