MKesenheimer/fault-injection-library #14: Errors in python scripts

lfldp1, 2025-03-12:
Unfortunately, there is no way to run the example script correctly. The command .\python pico-glitcher.py --rpico COM7 --delay 1000 1000 --length 100 100 returns the following errors:
 .\python pico-glitcher.py --rpico COM7 --delay 1000 1000 --length 100 100
Traceback (most recent call last):
  File "D:\INSTALKI\ANACONDA\pico-glitcher.py", line 162, in <module>
    main = Main(args)
           ^^^^^^^^^^
  File "D:\INSTALKI\ANACONDA\pico-glitcher.py", line 51, in __init__
    self.glitcher.init(port=args.rpico, ext_power=args.power, ext_power_voltage=3.3)
  File "C:\Users\dom\AppData\Roaming\Python\Python312\site-packages\findus\findus.py", line 672, in init
    self.pico_glitcher.init(port, 'PicoGlitcher')
  File "C:\Users\dom\AppData\Roaming\Python\Python312\site-packages\findus\findus.py", line 379, in init
    self.pyb.exec(f'import {micropy_script}')
  File "C:\Users\dom\AppData\Roaming\Python\Python312\site-packages\findus\pyboard.py", line 498, in exec_
    raise PyboardError("exception", ret, ret_err)
findus.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\n  File "PicoGlitcher.py", line 139, in <module>\r\nNameError: name \'MUX0_PIO_INIT\' isn\'t defined\r\n')
[+] Terminating gracefully.

PICO GLITCHER V1 has been reprogrammed with the v1 config file so it should work

MKesenheimer, 2025-03-12:
Hey, sorry for that. I will try to reproduce the error and fix it, if possible.

lfldp1, 2025-03-12:
> Hey, sorry for that. I will try to reproduce the error and fix it, if possible.

Ok thank you very much master :)

MKesenheimer, 2025-03-12:
Fixed in Version 1.3.1.

to update:
```
mkdir my-fi-project && cd my-fi-project
python -m venv .venv && source .venv/bin/activate
pip install findus==1.3.1
cd .venv/lib/python3.xy/site-packages/findus/firmware
upload --port /dev/tty.<rpi-tty-port> --files AD910X.py FastADC.py PicoGlitcher.py PulseGenerator.py Spline.py <config-path>/config.json
```

MKesenheimer, 2025-03-12:
Hmm, do you have the correct micropython firmware installed?

https://micropython.org/resources/firmware/RPI_PICO-20240602-v1.23.0.uf2

Also, I never tried the software on Windows. Do you have the possibility to execute on Linux, for example via an virtual machine?

lfldp1, 2025-03-12:
> Hmm, do you have the correct micropython firmware installed?
> 
> https://micropython.org/resources/firmware/RPI_PICO-20240602-v1.23.0.uf2
> 
> Also, I never tried the software on Windows. Do you have the possibility to execute on Linux, for example via a virtual machine?

Yes 1.23.0

lfldp1, 2025-03-12:
> Hmm, do you have the correct micropython firmware installed?
> 
> https://micropython.org/resources/firmware/RPI_PICO-20240602-v1.23.0.uf2
> 
> Also, I never tried the software on Windows. Do you have the possibility to execute on Linux, for example via a virtual machine?
> 
> Edit: I think I know what is happening here. Your python interpreter tries to import findus/firmware/PicoGlitcher which was never intended to import on the host system. But I am not sure why this happens. Maybe because you are using Windows, and things are a bit different there? I would have to investigate on Windows myself. Could you use Linux in the meantime?

!C:\Users\dom\AppData\Roaming\Python\Python312\Scripts\upload --port COM7 --files AD910X.py FastADC.py PicoGlitcher.py PulseGenerator.py Spline.py C:\Users\dom\AppData\Roaming\Python\Python312\site-packages\findus\firmware\config_v1\config.json

give this reprogramming result:

/AD910X.py
/FastADC.py
/PicoGlitcher.py
/PulseGenerator.py
/Spline.py
/config.json
/main.py
[+] Resetting Raspberry Pi Pico...
[+] Deleting AD910X.py...
[+] Uploading AD910X.py...
[+] Deleting FastADC.py...
[+] Uploading FastADC.py...
[+] Deleting PicoGlitcher.py...
[+] Uploading PicoGlitcher.py...
[+] Deleting PulseGenerator.py...
[+] Uploading PulseGenerator.py...
[+] Deleting Spline.py...
[+] Uploading Spline.py...
[+] Deleting config.json...
[+] Uploading C:\Users\dom\AppData\Roaming\Python\Python312\site-packages\findus\firmware\config_v1\config.json...
[+] Resetting Raspberry Pi Pico...
[+] Content of the Raspberry Pi Pico: but nothing received after : is this normall ?

lfldp1, 2025-03-12:
> Hmm, do you have the correct micropython firmware installed?
>
> https://micropython.org/resources/firmware/RPI_PICO-20240602-v1.23.0.uf2
>
> Also, I never tried the software on Windows. Do you have the [...]

yes I use windows. Tomorrow I will try to run it all on linux. By the way I will ask if these scripts are suitable to run in jupyter notebook?

MKesenheimer, 2025-03-12:
> !C:\Users\dom\AppData\Roaming\Python\Python312\Scripts\upload --port COM7 --files AD910X.py FastADC.py PicoGlitcher.py PulseGenerator.py Spline.py [...]

No, this is not normal. You see the content of your PicoGlitcher at the start of the output, which is the other way around. You can try uploading the files manually with Thonny (https://thonny.org).

lfldp1, 2025-03-12:
> > > Hmm, do you have the correct micropython firmware installed?
> > > https://micropython.org/resources/firmware/RPI_PICO-20240602-v1.23.0.uf2
> > > Also, I never tried the software on Windows. Do you have the possibility to execute on Linux, for example via a virtual machine?
> > > Edit: I think I know what is happening here. Your python interpreter tries to import findus/firmware/PicoGlitcher which was never intended to import on the host system. But I am not sure why this happens. Maybe because you are using Windows, and things are a bit different there? I would have to investigate on Windows myself. Could you use Linux in the meantime?
> > 
> > 
> > yes I use windows. Tomorrow I will try to run it all on linux. By the way I will ask if these scripts are suitable to run in jupyter notebook?
> 
> Ok, let me know how that goes.
> 
> Yes, the scripts should also work in a Jupyter notebook. But you have to convert the scripts manually to a notebook for better formatting. I am not a big fan of Jupyter, btw. However, if you want to covert the scripts into notebooks and like to open a pull request, I would include them in the repository.

Ok

lfldp1, 2025-03-12:
> Ok, let me know how that goes.
>
> Yes, the scripts should also work in a Jupyter notebook. But you have to convert the scripts manually to a notebook for better formatting. I am not a big fan of [...]

After installing everything on ubuntu, these errors occur:

(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware# upload --port /dev/ttyACM0 --files AD910X.py FastADC.py PicoGlitcher.py PulseGenerator.py Spline.py .venv/lib/python3.10/site-packages/findus/firmware/config_v1/config.json
[+] Resetting Raspberry Pi Pico...on
[+] Uploading AD910X.py...
[+] Uploading FastADC.py...
[+] Uploading PicoGlitcher.py...
[+] Uploading PulseGenerator.py...
[+] Uploading Spline.py...
[+] Deleting config.json...
[+] Uploading .venv/lib/python3.10/site-packages/findus/firmware/config_v1/config.json...

[+] Resetting Raspberry Pi Pico...
[+] Content of the Raspberry Pi Pico:

/AD910X.py
/FastADC.py
/PicoGlitcher.py
/PulseGenerator.py
/Spline.py
/config.json

(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware# ls
AD910X.py  config_v1  config_v2  decorators.py  FastADC.py  fault-injection-library.git  __init__.py  PicoGlitcher.py  PulseGenerator.py  __pycache__  Spline.py
(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware# wget https://github.com/MKesenheimer/fault-injection-library/archive/refs/tags/1.3.1.tar.gz
--2025-03-12 23:02:12--  https://github.com/MKesenheimer/fault-injection-library/archive/refs/tags/1.3.1.tar.gz
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/MKesenheimer/fault-injection-library/tar.gz/refs/tags/1.3.1 [following]
--2025-03-12 23:02:12--  https://codeload.github.com/MKesenheimer/fault-injection-library/tar.gz/refs/tags/1.3.1
Resolving codeload.github.com (codeload.github.com)... 140.82.121.9
Connecting to codeload.github.com (codeload.github.com)|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘1.3.1.tar.gz’

1.3.1.tar.gz                                           [                     <=>                                                                                         ]  95,05M  22,9MB/s    in 4,1s    

2025-03-12 23:02:17 (23,0 MB/s) - ‘1.3.1.tar.gz’ saved [99668465]

(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware# tar -xvf 1.3.1.tar.gz
fault-injection-library-1.3.1/
DID UNPACKED ALL FILES.......

(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware# ls
1.3.1.tar.gz  config_v1  decorators.py  fault-injection-library-1.3.1  __init__.py      PulseGenerator.py  Spline.py
AD910X.py     config_v2  FastADC.py     fault-injection-library.git    PicoGlitcher.py  __pycache__
(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware# cd fault-injection-library-1.3.1
(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware/fault-injection-library-1.3.1# ls
docs  examples  findus  LICENSE  mkdocs.yml  projects  pyproject.toml  README.md  requirements.txt  schematics  TODO.md
(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware/fault-injection-library-1.3.1# cd examples
(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware/fault-injection-library-1.3.1/examples# ls
CMD                            pico-glitcher-multiplexing-optimization.py  pico-glitcher-pulse-shaping-optimization.py  pico-glitcher.py               pro-glitcher.py
pico-glitcher-edge-counter.py  pico-glitcher-multiplexing.py               pico-glitcher-pulse-shaping.py               pico-glitcher-uart-trigger.py
(.venv) root@lukasz-virtual-machine:~/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware/fault-injection-library-1.3.1/examples# python3 pico-glitcher.py --rpico /dev/ttyACM0 --delay 1000 1000 --length 100 100
Traceback (most recent call last):
  File "/root/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware/fault-injection-library-1.3.1/examples/pico-glitcher.py", line 162, in <module>
    main = Main(args)
  File "/root/my-fi-project/.venv/lib/python3.10/site-packages/findus/firmware/fault-injection-library-1.3.1/examples/pico-glitcher.py", line 51, in __init__
    self.glitcher.init(port=args.rpico, ext_power=args.power, ext_power_voltage=3.3)
  File "/root/my-fi-project/.venv/lib/python3.10/site-packages/findus/findus.py", line 672, in init
    self.pico_glitcher.init(port, 'PicoGlitcher')
  File "/root/my-fi-project/.venv/lib/python3.10/site-packages/findus/findus.py", line 379, in init
    self.pyb.exec(f'import {micropy_script}')
  File "/root/my-fi-project/.venv/lib/python3.10/site-packages/findus/pyboard.py", line 498, in exec_
    raise PyboardError("exception", ret, ret_err)
findus.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\n  File "PicoGlitcher.py", line 24, in <module>\r\nOSError: [Errno 2] ENOENT\r\n')
[+] Terminating gracefully.
[... cut ...]
  File "<stdin>", line 1
    pico-glitcher.py --rpico /dev/ttyACM0 --delay 1000 1000 --length 100 100
                                                  ^^^^
SyntaxError: invalid syntax
>>> from findus import Database, PicoGlitcher
>>> pico-glitcher.py --rpico /dev/ttyACM0 --delay 1000 1000 --length 100 100
  File "<stdin>", line 1
    pico-glitcher.py --rpico /dev/ttyACM0 --delay 1000 1000 --length 100 100
                                                  ^^^^
SyntaxError: invalid syntax
>>> 

What am I doing wrong?

MKesenheimer, 2025-03-13:
Thanks. I don't understand what is happening here. It is also hardly readable. Could you please use code fences in the future?

\`\`\`
your code here
\`\`\`

MKesenheimer, 2025-03-13:
I tried again step by step with this commands:

```
cd ~
mkdir my-fi-project && cd my-fi-project
python3 -m venv .venv && source .venv/bin/activate
pip install findus
cd .venv/lib/python3.13/site-packages/findus/firmware
upload --port /dev/tty.usbmodem1101 --files AD910X.py FastADC.py PicoGlitcher.py PulseGenerator.py Spline.py config_v1/config.json
cd ~/my-fi-project
wget https://raw.githubusercontent.com/MKesenheimer/fault-injection-library/refs/heads/master/examples/pico-glitcher.py
python pico-glitcher.py --rpico /dev/tty.usbmodem1101 --delay 0 0 --length 100 100
```

Please try again with these commands so that we can compare our results.

lfldp1, 2025-03-13:
> I tried step by step with these commands and it worked in my case:
> 
> ```
> cd ~
> mkdir my-fi-project && cd my-fi-project
> python3 -m venv .venv && source .venv/bin/activate
> pip install findus
> cd .venv/lib/python3.13/site-packages/findus/firmware
> upload --port /dev/tty.usbmodem1101 --files AD910X.py FastADC.py PicoGlitcher.py PulseGenerator.py Spline.py config_v1/config.json
> cd ~/my-fi-project
> wget https://raw.githubusercontent.com/MKesenheimer/fault-injection-library/refs/heads/master/examples/pico-glitcher.py
> python pico-glitcher.py --rpico /dev/tty.usbmodem1101 --delay 0 0 --length 100 100
> ```
> 
> Please try again with these commands so that we can compare our results.

Hello We found the problem. It looks like installing pip install findus==1.3.1 was causing the problem. After entering the last commands pico glitcher starts:
`lukasz@lukasz-virtual-machine:~$ su
Password: 
root@lukasz-virtual-machine:/home/lukasz# cd
root@lukasz-virtual-machine:~# cd ~
root@lukasz-virtual-machine:~# mkdir my-fi-project && cd my-fi-project
mkdir: cannot create directory ‘my-fi-project’: File exists
root@lukasz-virtual-machine:~# mkdir my-fi-project1 && cd my-fi-project1
root@lukasz-virtual-machine:~/my-fi-project1# python3 -m venv .venv && source .venv/bin/activate
(.venv) root@lukasz-virtual-machine:~/my-fi-project1# pip install findus
Collecting findus
  Using cached findus-1.3.1-py3-none-any.whl (110 kB)
Collecting dash_ag_grid
  Using cached dash_ag_grid-31.3.0-py3-none-any.whl (3.9 MB)
Collecting dash_bootstrap_components
  Using cached dash_bootstrap_components-1.7.1-py3-none-any.whl (229 kB)
Collecting pandas
  Using cached pandas-2.2.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (13.1 MB)
Collecting scipy
  Using cached scipy-1.15.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (37.6 MB)
Requirement already satisfied: setuptools in ./.venv/lib/python3.10/site-packages (from findus) (59.6.0)
Collecting adafruit-ampy
  Using cached adafruit_ampy-1.1.0-py2.py3-none-any.whl (16 kB)
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting plotly
  Using cached plotly-6.0.0-py3-none-any.whl (14.8 MB)
Collecting dash
  Using cached dash-2.18.2-py3-none-any.whl (7.8 MB)
Collecting matplotlib
  Using cached matplotlib-3.10.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (8.6 MB)
Collecting click
  Using cached click-8.1.8-py3-none-any.whl (98 kB)
Collecting python-dotenv
  Using cached python_dotenv-1.0.1-py3-none-any.whl (19 kB)
Collecting requests
  Using cached requests-2.32.3-py3-none-any.whl (64 kB)
Collecting dash-core-components==2.0.0
  Using cached dash_core_components-2.0.0-py3-none-any.whl (3.8 kB)
Collecting nest-asyncio
  Using cached nest_asyncio-1.6.0-py3-none-any.whl (5.2 kB)
Collecting dash-html-components==2.0.0
  Using cached dash_html_components-2.0.0-py3-none-any.whl (4.1 kB)
Collecting retrying
  Using cached retrying-1.3.4-py3-none-any.whl (11 kB)
Collecting importlib-metadata
  Using cached importlib_metadata-8.6.1-py3-none-any.whl (26 kB)
Collecting typing-extensions>=4.1.1
  Using cached typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Collecting dash-table==5.0.0
  Using cached dash_table-5.0.0-py3-none-any.whl (3.9 kB)
Collecting Werkzeug<3.1
  Using cached werkzeug-3.0.6-py3-none-any.whl (227 kB)
Collecting Flask<3.1,>=1.0.4
[... 66 lines cut ...]
/FastADC.py
/PicoGlitcher.py
/PulseGenerator.py
/Spline.py
/config.json
(.venv) root@lukasz-virtual-machine:~/my-fi-project1/.venv/lib/python3.10/site-packages/findus/firmware# cd ~/my-fi-project1 
(.venv) root@lukasz-virtual-machine:~/my-fi-project1# wget https://raw.githubusercontent.com/MKesenheimer/fault-injection-library/refs/heads/master/examples/pico-glitcher.py
--2025-03-13 08:35:27--  https://raw.githubusercontent.com/MKesenheimer/fault-injection-library/refs/heads/master/examples/pico-glitcher.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7678 (7,5K) [text/plain]
Saving to: ‘pico-glitcher.py’

pico-glitcher.py                 100%[==========================================================>]   7,50K  --.-KB/s    in 0s      

2025-03-13 08:35:27 (26,5 MB/s) - ‘pico-glitcher.py’ saved [7678/7678]

(.venv) root@lukasz-virtual-machine:~/my-fi-project1# python pico-glitcher.py --rpico /dev/ttyACM0 --delay 0 0 --length 100 100
[+] Version of Pico Glitcher: [1, 3, 1]
[+] Version of findus: [1, 3, 1]
('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\n  File "PicoGlitcher.py", line 899, in block\r\nException: Function execution timed out!\r\n')
[-] Timeout received in block(). Continuing.
[+] Experiment 0	0	(0)	100	0	Y	b'Timeout'
`   This timeout means whats ?

MKesenheimer, 2025-03-13:
Please use code fences for longer pastes.

\`\`\`
your code here
\`\`\`

> This timeout means whats ?

You probably didn't connect RESET to TRIGGER. See here: https://fault-injection-library.readthedocs.io/en/latest/examples/

lfldp1, 2025-03-13:
> Please use code fences for longer pastes.

Please use code fences for longer pastes. - Yes and i did that in last post i use "code" button. I understand and thank for yours help will check this reset.

MKesenheimer, 2025-03-13:
> Please use code fences for longer pastes. - Yes and i did that in last post i use "code" button

For longer pastes over multiple lines the "code" button will not work. You will need to fence your pastes with three back-ticks:

\`\`\`
your code here
\`\`\`

lfldp1, 2025-03-13:
> > Please use code fences for longer pastes. - Yes and i did that in last post i use "code" button
> 
> For longer pastes over multiple lines the "code" button will not work. You will need to fence your pastes with three back-ticks:
> 
> ``` your code here ```

Ok i understand you now :)

closed 2025-03-14 by MKesenheimer