MKesenheimer/fault-injection-library #17: dead_time switch to ns

inode-, 2025-03-14:
Hello,

I was trying to use the dead_time and I got some troubles.

After checking the code I saw that the value is is seconds and not in nanoseconds like all others parameters.  Also in the documentation that is not specified.

`dead_time (float, default: 0 ) – Set a dead time that prohibits triggering within a certain time (trigger rejection). This is intended to exclude false trigger conditions. Can also be set to 0 to disable this feature.`

I think that it it's better to use the same scale on all parameters.

The code related to this issue is:

https://github.com/MKesenheimer/fault-injection-library/blob/e5f5ca0029e1c198d04d33ec58e0d2e32355fec9/findus/firmware/PicoGlitcher.py#L715

MKesenheimer, 2025-03-14:
Thanks. Yes, I am struggling also quite a bit with the inconsistent units. This is a future TODO.

MKesenheimer, 2025-06-27:
Changing the dead time from seconds to nanoseconds will likely break some examples and projects scripts. This is why I refused to implement nanoseconds everywhere so far. Also, python's sleep function is also in seconds which is why there is already a consistency with seconds in the code. I think, it might be ok to leave every time unit in seconds, except for length and delay. Therefore, unfortunately, I would like to refuse the change from seconds to nanoseconds. Maybe the documentation should be made clearer in some cases.