MKesenheimer/fault-injection-library #20: Script execution error.

Petr330, 2025-03-15:
Hello,
the code projects stm32f40x currently doesn't work.
```
>>> %FastDebug stm32f4-glitching.py --rpico /dev/ttyACM0 --target /dev/ttyUSB1 --delay 125000 125000 --length 20 30 --no-store
[+] Version of Pico Glitcher: [1, 3, 2]
[+] Version of findus: [1, 3, 2]
[+] Opening serial port /dev/ttyUSB1.
[-] Timeout received in block(). Continuing.
Traceback (most recent call last):
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching.py", line 165, in <module>
    main.run()
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching.py", line 129, in run
    speed = self.glitcher.get_speed(self.start_time, experiment_id)
AttributeError: 'Main' object has no attribute 'start_time'
>>>
```
Adding this code for "class Main:" fixes the error:
```
self.start_time = int(time.time())
```
But then another error occurs:
```
(.venv) user1@user1-Vostro-5502:~/fault-injection-library/projects/stm32f40x$ python stm32f4-glitching.py --rpico /dev/ttyACM0 --target /dev/ttyUSB1 --delay 125000 125000 --length 20 30 --no-store
[+] Version of Pico Glitcher: [1, 3, 2]
[+] Version of findus: [1, 3, 2]
[+] Opening serial port /dev/ttyUSB1.
[-] Timeout received in block(). Continuing.
[+] Experiment 0	0	(0)	125000	26	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 1	0	(0)	125000	23	O	b"b'warning: timeout'"
........................................................................................................................
[-] Timeout received in block(). Continuing.
[+] Experiment 30	0	(1)	125000	23	O	b"b'warning: timeout'"
Traceback (most recent call last):
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching.py", line 166, in <module>
    main.run()
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching.py", line 143, in run
    self.error_handler.check(experiment_id=experiment_id, response=response, expected=b'expected', user_action=error_action)
  File "/home/user1/fault-injection-library/.venv/lib/python3.12/site-packages/findus/findus.py", line 1238, in check
    parameters = (experiment_id - self.look_back + 1, ) + parameters[1:-2] + ('O', str(response).encode("utf-8"))
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
TypeError: can only concatenate tuple (not "list") to tuple
[+] Terminating gracefully.
[+] Closing serial port.
```
Could you check the code? Thanks in advance.

MKesenheimer, 2025-03-15:
Thanks, I will look into it.

MKesenheimer, 2025-03-15:
I could verify the first issue which I fixed in the latest push. You can update via `git pull`.

However I was not able to verify the second issue. Could you please go to `findus/findus.py` line `1238` and add these lines?

```
                    print(f"debug1 = {parameters[1:-2]}")
                    print(f"debug2 = {(experiment_id - self.look_back + 1, )}")
```

What is the output before the exception is displayed? Look out for the lines `debug1` and `debug2`. You have to install findus manually in this case:

```
cd fault-injection-library
pip install .
```

MKesenheimer, 2025-03-15:
Are you on Windows or Linux?

Petr330, 2025-03-15:
> Are you on Windows or Linux?

Yes.
```
(.venv) user1@user1-Vostro-5502:~/fault-injection-library$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
(.venv) user1@user1-Vostro-5502:~/fault-injection-library$ python -V
Python 3.12.3
(.venv) user1@user1-Vostro-5502:~/fault-injection-library$
```

Petr330, 2025-03-15:
> I could verify the first issue which I fixed in the latest push. You can update via `git pull`.
>
> However I was not able to verify the second issue. Could you please go to `findus/findus.py` line [...]

There is an error:
```
(.venv) user1@user1-Vostro-5502:~/fault-injection-library/projects/stm32f40x$ python stm32f4-glitching.py --rpico /dev/ttyACM0 --target /dev/ttyUSB1 --delay 125000 125000 --length 20 30 --no-store
[+] Version of Pico Glitcher: [1, 3, 2]
[+] Version of findus: [1, 3, 2]
[+] Opening serial port /dev/ttyUSB1.
[-] Timeout received in block(). Continuing.
[+] Experiment 0	0	(0)	125000	22	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 1	0	(0)	125000	24	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 2	0	(0)	125000	26	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 3	0	(0)	125000	25	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 4	0	(0)	125000	20	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 5	0	(0)	125000	21	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 6	0	(0)	125000	27	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 7	0	(0)	125000	24	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 8	0	(0)	125000	25	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 9	0	(0)	125000	27	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 10	0	(0)	125000	25	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 11	0	(0)	125000	30	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 12	0	(0)	125000	21	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 13	0	(0)	125000	25	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 14	0	(0)	125000	30	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 15	0	(0)	125000	30	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 16	0	(0)	125000	28	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 17	0	(0)	125000	28	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 18	0	(0)	125000	27	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 19	0	(0)	125000	30	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 20	0	(0)	125000	29	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 21	0	(0)	125000	25	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 22	0	(0)	125000	30	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 23	0	(0)	125000	22	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 24	0	(0)	125000	29	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 25	0	(0)	125000	24	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 26	0	(0)	125000	29	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 27	0	(0)	125000	27	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 28	0	(0)	125000	25	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 29	0	(0)	125000	20	O	b"b'warning: timeout'"
[-] Timeout received in block(). Continuing.
[+] Experiment 30	0	(0)	125000	24	O	b"b'warning: timeout'"
debug1 = []
debug2 = (1,)
Traceback (most recent call last):
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching.py", line 166, in <module>
    main.run()
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching.py", line 143, in run
    self.error_handler.check(experiment_id=experiment_id, response=response, expected=b'expected', user_action=error_action)
  File "/home/user1/fault-injection-library/.venv/lib/python3.12/site-packages/findus/findus.py", line 1240, in check
    parameters = (experiment_id - self.look_back + 1, ) + parameters[1:-2] + ('O', str(response).encode("utf-8"))
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
TypeError: can only concatenate tuple (not "list") to tuple
[+] Terminating gracefully.
[+] Closing serial port.
(.venv) user1@user1-Vostro-5502:~/fault-injection-library/projects/stm32f40x$
```

Petr330, 2025-03-15:
Figured it out a bit. This error occurs when using the --no-store key.

MKesenheimer, 2025-03-15:
ah, yes, that makes sense. Then, no experiments can be loaded from the database and error handling fails. Thanks for reporting, I will figure out a fix.

MKesenheimer, 2025-03-15:
Pushed another fix. I added a try except block which should be sufficient in that case.

closed 2025-03-15 by MKesenheimer