MKesenheimer/fault-injection-library #23: Error reading response bytes through Serial Port.

Petr330, 2025-03-19:
Hello.
I am working with the stm32f40x project.
The script has an error reading bytes via the serial port.
```
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ python stm32f4-glitching.py --rpico /dev/ttyACM0 --target /dev/ttyUSB0 --delay 125000 125000 --length 20 30
[+] Version of Pico Glitcher: [1, 5, 0]
[+] Version of findus: [1, 5, 0]
[+] Opening serial port /dev/ttyUSB0.
UARTsetPort = Serial<id=0x713f87722110, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='E', stopbits=1, timeout=1, xonxoff=False, rtscts=False, dsrdtr=False)
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 0	0	(0)	125000	23	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'\xb9'
[-] Timeout received in block(). Continuing.
[+] Experiment 1	0	(0)	125000	22	O	b'warning: timeout'
init_bootloader_read = b'\xf9'
[+] Experiment 2	0	(1)	125000	26	M	b'error: bootloader error'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 3	0	(1)	125000	22	O	b'warning: timeout'
init_bootloader_read = b'\xf9'
[+] Experiment 4	0	(1)	125000	20	M	b'error: bootloader error'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 5	0	(1)	125000	29	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 6	0	(1)	125000	21	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 7	0	(1)	125000	22	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 8	0	(1)	125000	21	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'\xb9'
[-] Timeout received in block(). Continuing.
[+] Experiment 9	0	(1)	125000	22	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 10	0	(1)	125000	29	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 11	0	(1)	125000	27	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 12	0	(1)	125000	25	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 13	0	(1)	125000	20	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 14	0	(1)	125000	24	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 15	0	(1)	125000	25	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 16	0	(1)	125000	28	O	b'warning: timeout'
init_bootloader_read = b'\xf9'
[+] Experiment 17	0	(1)	125000	30	M	b'error: bootloader error'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 18	0	(1)	125000	30	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 19	0	(1)	125000	25	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 20	0	(1)	125000	25	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 21	0	(1)	125000	21	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 22	0	(1)	125000	26	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 23	0	(1)	125000	29	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 24	0	(1)	125000	20	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 25	0	(1)	125000	26	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 26	0	(1)	125000	21	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 27	0	(1)	125000	26	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 28	0	(1)	125000	30	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 29	0	(1)	125000	24	O	b'warning: timeout'
init_bootloader_read = b'y'
setup_memread_read = b'y'
[-] Timeout received in block(). Continuing.
[+] Experiment 30	0	(1)	125000	25	O	b'warning: timeout'
^C
Exitting...
[+] Terminating gracefully.
[+] Closing serial port.
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ 

```
For more information, I added data output to the script.

```
 print(f"[+] Opening serial port {port}.")
        self.ser = serial.Serial(port=port, baudrate=115200, timeout=1, bytesize=8, parity="E", stopbits=1)
        print(f"UARTsetPort = {self.ser}")
```
```
    # init bootloader
        self.ser.write(b'\x7f')
        s = self.ser.read(1)
        print(f"init_bootloader_read = {s}")
        if s == self.ACK:
            return b'ok: bootloader ok'
        return b'error: bootloader error'
```
```
 # read memory command (x11: read memory, xee: crc)
        self.ser.write(b'\x11\xee')
        if read:
            s = self.ser.read(1)
            print(f"setup_memread_read = {s}")
```
At the same time, the logical analyzer shows correct data exchange. Wireshark interception also shows correct data exchange.

[image]

[image]

[image]

Petr330, 2025-03-19:
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Python 3.12.3

MKesenheimer, 2025-03-19:
What do you mean by "the script has an error reading the bytes"? Are you worried about the timeout or the error

```
...
init_bootloader_read = b'\xf9'
[+] Experiment 17	0	(1)	125000	30	M	b'error: bootloader error'
...
```

Maybe you can also try the project `stm32f42x` which should be nearly identical.

Petr330, 2025-03-19:
My target device is in RDP0 mode.

MKesenheimer, 2025-03-19:
Ah understood. Can you please print the variable `response` at each step and run a few more experiments?

MKesenheimer, 2025-03-19:
I know what's wrong. Sorry for that. I push a solution in the next few minutes.

MKesenheimer, 2025-03-19:
Ok, after a few attempts, it should work now.

Petr330, 2025-03-19:
Sorry, I probably misled you. Error reading bytes occurs in the STM32Bootloader.py script when running the stm32f4-glitching.py script. The script STM32Bootloader.py does not receive the correct data for its work. I made the changes indicated above to the STM32Bootloader.py script.

MKesenheimer, 2025-03-19:
Alright, can you still please print the variable `response` at each step and run a few more experiments?

MKesenheimer, 2025-03-19:
I quickly verified with my setup. It seems to work on my side.

Petr330, 2025-03-19:
> Хорошо, можете ли вы все равно распечатать переменную `response`на каждом шаге и провести еще несколько экспериментов?
In which script? I think the problem is in the STM32Bootloader.py script when reading data from the serial port. Now two functions are running in a cycle - bootloader initialization and memory read attempt. The script cannot get correct data from the target board.

MKesenheimer, 2025-03-19:
I verified with the script `fault-injection-library/projects/stm32f40x/stm32f4-glitching.py` from the latest push. Maybe something else is wrong? Did you connect the trigger input to the UART-TX line?

Please post also an output of the variable `response` at each step of the script `stm32f4-glitching.py`.

Petr330, 2025-03-19:
Tomorrow I will be able to conduct experiments. Everything is connected correctly. stm32flash works fine via the serial port interface.
```
user1@user1-ub24:~/stm32flash-code$ stm32flash -b 115200 /dev/ttyUSB0
stm32flash 0.7

http://stm32flash.sourceforge.net/

Interface serial_posix: 115200 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0413 (STM32F40xxx/41xxx)
- RAM        : Up to 128KiB  (12288b reserved by bootloader)
- Flash      : Up to 1024KiB (size first sector: 1x16384)
- Option bytes  : 16b
- System memory : 30KiB
```

MKesenheimer, 2025-03-19:
ok, perfect. Let me know how it goes. I also tried to optimize the script a bit. Now the serial timeouts are a bit better and the script should be faster.

Petr330, 2025-03-19:
What version of python are you using?

MKesenheimer, 2025-03-19:
Python 3.13.2

Should not make a huge difference.

MKesenheimer, 2025-03-19:
Remember that you probably have to play with the timeouts at your side too. The stm32f4 is a bit picky in that regard.

Petr330, 2025-03-20:
I updated the project. Added the output of response data to the stm32f4-glitching.py script. I ran tests. Here are the results:
```
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ python stm32f4-glitching_m.py --rpico /dev/ttyACM0 --target /dev/ttyUSB0 --delay 125000 125000 --length 20 30
[+] Version of Pico Glitcher: [1, 5, 0]
[+] Version of findus: [1, 5, 0]
[+] Opening serial port /dev/ttyUSB0.
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 0	0	(0)	125000	26	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 1	0	(0)	125000	23	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 2	0	(0)	125000	20	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 3	0	(0)	125000	22	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 4	0	(0)	125000	20	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 5	0	(0)	125000	27	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 6	0	(0)	125000	28	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 7	0	(0)	125000	29	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 8	0	(0)	125000	28	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 9	0	(0)	125000	27	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 10	0	(0)	125000	21	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 11	0	(0)	125000	26	O	b'warning: timeout'
response_line97_init_bootloader = b'error: bootloader error'
response_line127_color = b'error: bootloader error'
[+] Experiment 12	0	(0)	125000	25	M	b'error: bootloader error'
response_line97_init_bootloader = b'ok: bootloader ok'
[... 100 lines cut ...]
Info : [stm32f4x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32f4x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x1fff3da0 msp: 0x20001d80
shutdown command invoked

response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
response_line127_color = b'warning: timeout'
[+] Experiment 31	0	(0)	125000	24	O	b'warning: timeout'
response_line97_init_bootloader = b'ok: bootloader ok'
response_line104_setup_memread = b'success: RDP inactive'
[-] Timeout received in block(). Continuing.
^C
Exitting...
[+] Terminating gracefully.
[+] Closing serial port.
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ 
```
The script initializes the bootloader and checks the RDP.
For some reason the script does not proceed to the process of reading memory.

MKesenheimer, 2025-03-20:
Ok, understood. This is because the timeout condition overwrites any response from the target. The script goes into timeout because the trigger condition is not met. For testing, you can either comment the line `self.glitcher.block(timeout=1)` or connect the TX line of your uart adapter to the trigger input of the Pico Glitcher (so that the Pico Glitcher can observe the trigger condition).

MKesenheimer, 2025-03-20:
You can also try communicating with your target with the command-line tool `power-cycle-and-read`:
https://fault-injection-library.readthedocs.io/en/latest/cli/

MKesenheimer, 2025-03-24:
Hey, good to hear! Yes that rate is acceptable. We have to glitch for every read attempt and getting the glitch position right is hard. You can of course try to optimize the glitch length and delay. Maybe it can be improved a little.

I am using macos and linux for running these scripts.

Thanks for reporting this issue.

closed 2025-03-24 by MKesenheimer