MKesenheimer/fault-injection-library #25: Error updating database.

Petr330, 2025-03-24:
After updating the project, the database server starts, but when updating the database, an error occurs:
```
Traceback (most recent call last):
  File "/home/user1/fault-injection-library/.venv/lib/python3.12/site-packages/findus/analyzer/analyzer.py", line 158, in update_dropdown
    return get_databases(DATABASE_DIRECTORY)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user1/fault-injection-library/.venv/lib/python3.12/site-packages/findus/analyzer/analyzer.py", line 81, in get_databases
    databases_new.append('%s (%d)' %(databases[index], get_number_of_experiments(directory, databases[index])))
                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user1/fault-injection-library/.venv/lib/python3.12/site-packages/findus/analyzer/analyzer.py", line 34, in get_number_of_experiments
    cursor.execute(query)
sqlite3.OperationalError: attempt to write a readonly database
```

MKesenheimer, 2025-03-24:
What was your command? Did you use `resume`?

Petr330, 2025-03-24:
I used the stm32f4-glitching.py script command with the --resume key and without it. There are many database files in the database directory, but the server does not work with them.

[image]
DB Browser works fine with these files.

```
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ analyzer --directory databases
Dash is running on http://127.0.0.1:8080/

 * Serving Flask app 'findus.analyzer.analyzer'
 * Debug mode: on
```
When opened in a web browser, the database is not updated.
[image]

MKesenheimer, 2025-03-24:
Weird. Did you change the number of parameters in between? Can you please delete or rename the database folder and try again with a completely new setup?

Petr330, 2025-03-24:
I cleared the database directory, but when I run the script I get an error:
```
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ python stm32f4-glitching_m_v2.py --rpico /dev/ttyACM0 --target /dev/ttyUSB0 --delay 108000 108000 --length 1000 1000 --resume
[+] Version of Pico Glitcher: [1, 5, 0]
[+] Version of findus: [1, 5, 0]
Traceback (most recent call last):
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching_m_v2.py", line 170, in <module>
    main = Main(args)
           ^^^^^^^^^^
  File "/home/user1/fault-injection-library/projects/stm32f40x/stm32f4-glitching_m_v2.py", line 60, in __init__
    self.database = Database(sys.argv, resume=self.args.resume, nostore=self.args.no_store)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user1/fault-injection-library/.venv/lib/python3.12/site-packages/findus/findus.py", line 87, in __init__
    latest_file = max(list_of_files, key=os.path.getctime)[10:]
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: max() iterable argument is empty
[+] Terminating gracefully.
(.venv) user1@user1-ub24:~/fault-injection-library/projects/stm32f40x$ 
```

Petr330, 2025-03-24:
Yes, the database is working. First we clear the directory, run the script once without the --resume key, then with the key and the database files are visible through the browser.

MKesenheimer, 2025-03-24:
Understood. I will clarify the --resume argument and make it easier to use. Is now everything working as expected?

Petr330, 2025-03-24:
Yes, thank you.

closed 2025-03-25 by MKesenheimer