MKesenheimer/fault-injection-library #29: Backward compatybility

Flybro, 2025-06-22:
Hi,
I'm trying to run Pico Glitcher in hardware version 2.1 with STM8s as target with latest version of findus but something just doesn't want to work. And by the way, how backward is the findus library compatible with the hardware? I've also tried with the latest version (v1.11.2) but I get a message about version incompatibility but without any additional information.

Even when Pico, Glitcher, and findus are in the same version, errors still occur.

Installation:

    pip install findus==1.5.0
    git clone --branch 1.5.0 --single-branch https://github.com/MKesenheimer/fault-injection-library.git findus

Input: 
   
    python stm8-readmemory.py \
    --target /dev/ttyACM0 \
    --rpico /dev/ttyACM1 \
    --delay 106_000 107_500 \
    --length 10 100

Output:

    [+] Version of Pico Glitcher: [1, 5, 0]
    [+] Version of findus: [1, 5, 0]
    Traceback (most recent call last):
      File "/home/masterchief/Hardware/findus/projects/stm8s/stm8-readmemory.py", line 149, in <module>
        main = Main(args)
      File "/home/masterchief/Hardware/findus/projects/stm8s/stm8-readmemory.py", line 48, in __init__
        self.glitcher.init(port=args.rpico)
      File "/home/masterchief/.findus/lib/python3.10/site-packages/findus/findus.py", line 712, in init
        self.pico_glitcher.enable_vtarget()
      File "/home/masterchief/.findus/lib/python3.10/site-packages/findus/findus.py", line 450, in enable_vtarget
        self.pyb.exec(f'mp.enable_vtarget({use_mux})')
      File "/home/masterchief/.findus/lib/python3.10/site-packages/findus/pyboard.py", line 498, in exec_
        raise PyboardError("exception", ret, ret_err)
    findus.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nTypeError: function takes 1 positional arguments but 2 were given\r\n')
    [+] Terminating gracefully.

MKesenheimer, 2025-06-22:
Hey,

thanks for reporting this issue. The software is backwards compatible with all versions of the Pico Glitcher. For sure with all versions 2.x. Can you please try to use the latest version?

```
mkdir my-fi-project && cd my-fi-project
python3 -m venv .venv && source .venv/bin/activate
pip install findus==1.11.2
cd .venv/lib/python3.xx/site-packages/findus/firmware
upload --port /dev/tty.<rpi-tty-port> --files AD910X.py FastADC.py Globals.py \ 
    PicoGlitcher.py PulseGenerator.py Spline.py config_v2.1-2/config.json
```

Flybro, 2025-06-29:
Installing findus with the version specified helped, thanks.

MKesenheimer, 2025-06-29:
Perfect! Glad to help!

closed 2025-06-29 by MKesenheimer